Dukweeno/Duckuino #3: A few issues

d4n5h, 2016-09-29:
1. You can't use PHP on GitHub.

2. STRING command has a bug,it adds lead whites-spaces all the time.
Sometimes you need to add a white-space in the beginning of a string, but not all the time.
In my compiler you can do "STRING Without white space" and also "STRING  With white space".

3. Look at my updated GUI/Index page. i added Codemirror for syntax highlighting and and i've "updated" the design a bit.

Thecakeisgit, 2016-09-29:
1. Thanks i did'nt know.
2. Ill fix that
3. Cool, it's very good :D

d4n5h, 2016-09-29:
@Thecakeisgit 
I've reimplemented console debugging like in your compiler...
Now you don't have any excuse for not switching to my compiler :)

d4n5h, 2016-09-29:
@Thecakeisgit 
I've "Download Payload" button...

d4n5h, 2016-09-29:
@Thecakeisgit 
You didn't add REPLAY/REPEAT

d4n5h, 2016-09-29:
"CTRL ALT TEST" Don't work lol

Thecakeisgit, 2016-09-29:
Yep i know ^^ i'll add id this soon, i added payload dl too
Yes it works ! it throws an error as expected ! ;D

d4n5h, 2016-09-29:
Sure... if you don't realise that REPEAT is dependant on everything to be in same line, so my solution is better ;)

Thecakeisgit, 2016-09-29:
Leave me this ;D

Thecakeisgit, 2016-09-29:
You'll see :D

d4n5h, 2016-09-29:
@Thecakeisgit 
Issue with REPEAT/REPLAY:
Try doing:
CTRL a
REPEAT 10

Thecakeisgit, 2016-09-29:
Yes ?

d4n5h, 2016-09-30:
@Thecakeisgit 
Try:
CRTL ALT a
REPLAY 10

d4n5h, 2016-09-30:
@Thecakeisgit 
My bad, REPEAT is working great.
Can i use your solution?

closed 2016-09-30 by d4n5h

Thecakeisgit, 2016-09-30:
Of course you can do it ! ;D