Dukweeno/Duckuino #7: Your compiler is totally bugged

d4n5h, 2016-10-18:
1. You added the typeKey utility function in the end of the payload so it will not run...
Fix: Add it to the beginning of the payload/script (before the `void setup(){...};`).

2. You don't have support for PAGEUP/PAGEDOWN.

3. You didn't add `#include <Keyboard.h>` and/or `#include <Mouse.h>` at the beginning of the payload.

As it seems, you probably don't even have a atmega32u4 board to test your script.

I've fixed all of the above in my repository.

Thecakeisgit, 2016-10-18:
1. Ok Thanks, i'll fix it.
2. I'll add this too
3. Keyboard.h is already included by arduino core, why include it again ? c:

I got a wonderful arduino mini to test this ;D And i worked without the include ^^

d4n5h, 2016-10-18:
The Arduino (Pro) Mini is a ATmega328 based MCU.
You cannot use the Keyboard.h library with it so there is no way that you've tested it and it worked.

Thecakeisgit, 2016-10-18:
Sorry it was an Arduino Micro ;D

d4n5h, 2016-10-18:
Also, you would get a error because of the typeKey() function.

Thecakeisgit, 2016-10-18:
Nope, i tested ^^ No error.
With this:
```c
/*
 * Generated with <3 by Dckuino.js, an open source project !
 */

// Init function
void setup()
{
  // Begining the stream
  Keyboard.begin();

  // Waiting 500ms for init
  delay(500);

  typeKey(108);

  Keyboard.print("My name is");
}

void typeKey(int key)
{
  Keyboard.press(key);
  delay(50);
  Keyboard.release(key);
}

// Unused
void loop() {}
```

d4n5h, 2016-10-18:
It outputs an error.

You didn't even check it.

d4n5h, 2016-10-18:
```
Arduino: 1.6.11 (Windows 10), Board: "Arduino/Genuino Micro"

WARNING: Category '' in library UIPEthernet is not valid. Setting to 'Uncategorized'
C:\Users\Daniel\Documents\Arduino\sketch_oct18c\sketch_oct18c.ino: In function 'void setup()':

sketch_oct18c:9: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
   Keyboard.begin();

   ^

C:\Users\Daniel\Documents\Arduino\sketch_oct18c\sketch_oct18c.ino: In function 'void typeKey(int)':

sketch_oct18c:21: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
   Keyboard.press(key);

   ^

exit status 1
'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

d4n5h, 2016-10-18:
Again, you just bullshiting.... 
Just fix you shit....

Thecakeisgit, 2016-10-18:
Yes i'll fix it tomorrow, i've tested, but on linux, so i don't know what can change but i did'nt got any error *-*

d4n5h, 2016-10-18:
I've tested it on macOS and on Windows 10 and it's totally bugged...

Thecakeisgit, 2016-10-18:
Ok i'll fix i said, don't get angry..

Thecakeisgit, 2016-10-19:
When i add #include <Keyboard.h>, this happens : 
sketch_oct19a.ino:5:22: fatal error: Keyboard.h: No such file or directory
compilation terminated.

Thecakeisgit, 2016-10-19:
What's your arduino ide version ? Mine is 2:1.0.5+dfsg2-4

d4n5h, 2016-10-19:
It's in the error report - Arduino: 1.6.11

d4n5h, 2016-10-19:
I know what is the problem. You probably didn't set the MCU type to Arduino/Genuino Micro.
It's in Tools->Board.

Thecakeisgit, 2016-10-19:
It was set to Arduino Micro, and i compiled, without uploading and it did this error.. Kinda weird.

Thecakeisgit, 2016-10-19:
When i use Keyboard object, without including, this works.. \*-\*

Thecakeisgit, 2016-10-19:
My bad, i updated my arduino ide version and it works..

d4n5h, 2016-10-19:
Maybe we should add a note at the README file that indicates that the IDE must be updated to the latest version so there will be no cross-platform compatibility issues.

Thecakeisgit, 2016-10-19:
Yep ^^ it was a problem in Ubuntu repository, there is only arduino 2:1.0.5

Thecakeisgit, 2016-10-19:
[a7330f6](https://github.com/Thecakeisgit/Dckuino.js/commit/a7330f65e555fa80d547b88961cee64c65dbd79c) resolved it.

closed 2016-10-19 by Thecakeisgit