Dukweeno/Duckuino #10: Keyboard Library in other languages?

Pedrinbeep, 2016-10-25:
Hi Thecakeisgit!

You Dckuino.js is awesome but I have a little problem. My keyboard is not english and this makes mistakes when the script is runing. Do you have another library for a diferent keyboard? In my case is a Spanish keyboard. 

Thank you for your work.

Thecakeisgit, 2016-10-25:
Hey man :D, this can't be archived with a library, i got the same issue, because i got a french keyboard.. You need to modify your Arduino source code, the file HID.cpp contains a var named _asciimap.. You can search on the net to find the HID.cpp for your needs :)

Also, you gave me an idea to fix your problem, so i'll don't close the issue now. I'll inform u if i find something ;D

Thecakeisgit, 2016-10-25:
@Pedrinbeep

Thecakeisgit, 2016-10-25:
@Pedrinbeep did u find something ? @d4n5h Any tip ? :D

d4n5h, 2016-10-25:
@Pedrinbeep @Thecakeisgit
It's on my TODO list as well.
I think that the standard Arduino keyboard library is based on ASCII characters.
In order to bring new layouts we need somehow to switch to UTF-8 and add those characters to the compiler's array of keys.
I'll take a look on it later.

Thecakeisgit, 2016-10-25:
@d4n5h If u understand french http://www.zem.fr/utiliser-mouse-keyboard-azerty-arduino-pro-micro-teensy/

d4n5h, 2016-10-25:
@Thecakeisgit Not sure if it will work in this case but a simple workaround could be just to switch the keyboard input language to english with a "SHIFT ALT" combo. But if you have to print characters in other languages it will not work.

d4n5h, 2016-10-25:
@Thecakeisgit  Oh, i didn't knew that you have a totally different physical key layout. Weird...
So my "workaround" will not work in this case...

Thecakeisgit, 2016-10-25:
Yep, i know but, it don't work on every computer, for me i uninstalled english locale, so SHIFT ALT just do NOTHING, so that's not acceptable.. But we can create a "LocaleKeyboard" project that we use both.
The idea is to create modules(asciimap_fr, asciimap_es, ...) and to just do Keyboard.init("fr_FR") to set locale.. 
What do u say about that ? @d4n5h

d4n5h, 2016-10-25:
@Thecakeisgit Good idea! i'm in.

Thecakeisgit, 2016-10-25:
Add me as contributor ;D

Thecakeisgit, 2016-10-25:
@d4n5h

d4n5h, 2016-10-25:
In my repo?

d4n5h, 2016-10-25:
@Thecakeisgit I've added you as a collaborator.

Thecakeisgit, 2016-10-25:
@d4n5h Come :D https://github.com/Thecakeisgit/LocaleKeyboard

Thecakeisgit, 2016-10-25:
@d4n5h https://hack.chat/?LocaleKeyboard

Thecakeisgit, 2016-10-25:
@Pedrinbeep We're working on a port of Keyboard arduino Lib, here -> https://github.com/Thecakeisgit/LocaleKeyboard
Feel free to comment and help :+1:

Pedrinbeep, 2016-10-25:
@Thecakeisgit Great! I´ll try to help ;)

Pedrinbeep, 2016-10-26:
@Thecakeisgit @d4n5h I found this link, you can be helpful?
LINK: https://forums.hak5.org/index.php?/topic/36068-keyboard-layout-modified-spanish-keyboard-layout/

Thecakeisgit, 2016-10-26:
It could be helpful, if the links were not dead :/

Thecakeisgit, 2016-10-26:
@Pedrinbeep

Pedrinbeep, 2016-10-26:
I sent a few PM to this guys, if they response, I will let you know.
@Thecakeisgit

Pedrinbeep, 2016-10-26:
@Thecakeisgit Here is the es.properties file for download:
https://www.dropbox.com/s/4ibsw9l9m1y560r/es.properties?dl=0

I hope could help! ;)

Thecakeisgit, 2016-10-26:
@Pedrinbeep Thanks man, i got the solution, gimme tomorrow and it will be done ;D

Thecakeisgit, 2016-11-02:
Fixed with 7ee4217 & 02fd692.
Now we need to update in LocaleKeyboard.js.

closed 2016-11-02 by Thecakeisgit