Dukweeno/Duckuino #12: Enter problem

NovaHowl, 2016-11-17:
When using the Enter command, it just types (

Thecakeisgit, 2016-11-17:
Hi @NovaHowl, please tell me what is you keyboard layout, it's caused by your layout..

NovaHowl, 2016-11-17:
Australian

NovaHowl, 2016-11-17:
Everything else works fine, just the enter command,

NovaHowl, 2016-11-17:
Thecakeisgit

Thecakeisgit, 2016-11-17:
*-* It's weird, @NovaHowl are u sure your Keyboard layout is exactly the same as US Keyboard layout ?

NovaHowl, 2016-11-17:
Yep, and I tried switching it to UK and it didn't work.

Thecakeisgit, 2016-11-17:
Hmm okay, and you used <b>Download Sketch!</b> button ?

Thecakeisgit, 2016-11-17:
@NovaHowl

NovaHowl, 2016-11-17:
Download sketch doesn't work for me. I also noticed I had to remove #include "Keyboard.h" @Thecakeisgit

NovaHowl, 2016-11-17:
Otherwise it would say (error compiling for Teensy 3.1/3.2

Thecakeisgit, 2016-11-17:
WTF, Hmm what's ur Arduino ide version ?

Thecakeisgit, 2016-11-17:
Are you on linux ? @NovaHowl

NovaHowl, 2016-11-17:
1.6.11 @Thecakeisgit

NovaHowl, 2016-11-17:
Nope

Thecakeisgit, 2016-11-17:
Okay, same version, @NovaHowl so it's maybe because you're trying to compile an Arduino code based on Arduino core for teensy.. Don't u think ?

NovaHowl, 2016-11-17:
Yeah maybe... Everything else in the converter works just fine tho

NovaHowl, 2016-11-17:
GUI r works, STRING works and everything else, but the ENTER function doesn't work. @Thecakeisgit

Thecakeisgit, 2016-11-17:
Okay, because with same thing with an Arduino mini, it works perfectly well.. But why the #include "Keyboard.h" fails... I don't know, it could be this..

NovaHowl, 2016-11-17:
When leaving #include "Keyboard.h"

In file included from C:\Users\Nova\AppData\Local\Temp\arduino_modified_sketch_60053\sketch_nov18a.ino:5:0:

C:\Program Files (x86)\Arduino\libraries\Keyboard\src/Keyboard.h:25:17: fatal error: HID.h: No such file or directory

 #include "HID.h"

                 ^

compilation terminated.

Error compiling for board Teensy 3.2 / 3.1.

NovaHowl, 2016-11-17:
@Thecakeisgit

Thecakeisgit, 2016-11-17:
Hmm, yes it's definitely because it's a teensy.. I'm sorry, but i think i can't help you @NovaHowl because i don't know teensy libs.. Sorry.. But if you find any solution, please tell me, I'll be interested.. ;D

Do you mind if i close the issue ?

NovaHowl, 2016-11-17:
Alright, that's fine. This is the only converter that came close to working with my Teensy 3.2. Anyways...

Thecakeisgit, 2016-11-17:
I'm sorry man, good night ;D @NovaHowl

closed 2016-11-17 by Thecakeisgit

NovaHowl, 2016-11-17:
Good night to you, good morning to me ;)

crinny, 2018-08-19:
Actually same.

Nurrl, 2018-08-19:
Hi @crinny, what is your Locale ?

crinny, 2018-08-19:
Russian

JI8, 2018-12-12:
Is there a fix for this problem? ''typeKey(KEY_RETURN);'' returns '')''.

Nurrl, 2018-12-12:
Again, you need to select a locale to be bundled with your code for it to work :) @JI8

kenoc1, 2020-05-18:
Hey, I have the same problem. ''typeKey(KEY_RETURN);'' returns '')'' What do you mean with select a locale to be bundled with your code @Nurrl ?

Nurrl, 2020-05-19:
I mean use this:
[image]

Nurrl, 2020-05-19:
And select your locale, then try again to see if it works @kenoc1