Dukweeno/Duckuino #13: Layout messed up using Firefox/Chrome

gh7531, 2016-11-22:
Hello,
I tried to compile and download a script using the live version but if I load the page in Firefox or Chrome the layout of the page is messed up.
[image]

If I switch my useragent to a mobile device it looks fine.
[image]

After I changed the useragent to default is get messed up again.

Thecakeisgit, 2016-11-22:
I'll try to fix this, but i can't reproduce the bug because i got a 4:3 screen.. Could you try on dev.cak3repo.xyz(Who is my test server) @gh7531 ? I tried a fix :)

gh7531, 2016-11-22:
Nope isn't working :( it looks the same.
[image]

On my 4:3 second screen it's messed up to.

Thecakeisgit, 2016-11-22:
Hmm WTF, i can't reproduce it \*-\* @gh7531, have you updated your Firefox ?

Thecakeisgit, 2016-11-22:
Maybe your version doesn't support the css3 calc() function. @gh7531

Thecakeisgit, 2016-11-22:
What's you resolution ? @gh7531

Thecakeisgit, 2016-11-22:
Please retry @gh7531 https://dev.cak3repo.xyz

gh7531, 2016-11-22:
Still no luck :( 
I'm using FF 50.0 my resolution is 1920x1080.

While using the edge browser it's still messed up but not as badly
[image]

Thecakeisgit, 2016-11-22:
Okay, i'll wait to somebody else to send a comment here to really fix it \*-\* because i don't know what the fuck this is.. @gh7531 
@d4n5h Any idea ?

ImaFridge, 2016-11-23:
I've got the same problem as @gh7531 derp derp

Thecakeisgit, 2016-11-24:
Ok i'll fix it soon, i reinstalled my test server.. @gh7531 @ImaFridge

KevinDenys, 2016-11-25:
1920x1080 here running last version of chrome. No problems.

Thecakeisgit, 2016-11-25:
@KevinDenys I got no idea of what happenig with these guys.. I can't reproduce it, and i don't even know how to fix.. Any idea ? @KevinDenys

KevinDenys, 2016-11-25:
Can we get a screenshot of the console when the page is loaded @imafridge @gh7531

gh7531, 2016-11-26:
I'm not at home for the next 10 days so I can't provide the information I'm sorry. I'll do it as soon as I get home again.

ImaFridge, 2016-11-26:
[image]

Thecakeisgit, 2016-11-26:
Kinda weird, your Javascript console is saying something ? @ImaFridge

KevinDenys, 2016-11-26:
@ImaFridge thats the seytonic version. Can you try the https://thecakeisgit.github.io/Dckuino.js/

KevinDenys, 2016-11-26:
[image]
Firefox 1920x1080

Thecakeisgit, 2016-11-26:
572087d seems to fix it.. Try again please @ImaFridge :)

ImaFridge, 2016-11-26:
It works now! you guys rock <3
I used the Seytonic version bc i couldnt reacht the other site...

Thecakeisgit, 2016-11-26:
Thanks ! Fixed with 572087d.

closed 2016-11-26 by Thecakeisgit