Dukweeno/Duckuino #15: Keyboard layout

KevinDenys, 2016-11-25:
I have an Belgian Dot layout as keyboard. So when I try to write "azerty" using the programmed arduino leonardo it types "qwerty". How Can I fix this?

Thecakeisgit, 2016-11-25:
Approximately the same thing as #10

Thecakeisgit, 2016-11-25:
#10

Thecakeisgit, 2016-11-25:
You can use the **Download Sketch !** button ;D And select **fr_FR** locale, if you got a ~French Belgian keyboard ;D

KevinDenys, 2016-11-25:
Okay this fixed one of my problems, now I can't get Keyboard.press(KEY_LEFT_GUI); to work?

Thecakeisgit, 2016-11-25:
KEY_LEFT_GUI is hard coded, so if ut doesn't work, i don't know how you can make it work !
Try to post your entire code please ;D

KevinDenys, 2016-11-25:
`/*
 * Generated with <3 by Dckuino.js, an open source project !
 */

#include "Keyboard.h"

void typeKey(uint8_t key)
{
  Keyboard.press(key);
  delay(50);
  Keyboard.release(key);
}

/* Init function */
void setup()
{
  // Begining the Keyboard stream
  Keyboard.begin();

  // Wait 500ms
  delay(500);

  delay(750);

  Keyboard.press(KEY_LEFT_GUI);
  Keyboard.press('r');
  Keyboard.releaseAll();

  delay(1000);

  Keyboard.print("cmd");

  typeKey(KEY_RETURN);

  // Ending stream
  Keyboard.end();
}

/* Unused endless loop */
void loop() {}
`

Thecakeisgit, 2016-11-25:
Hmm kinda weird, i don't know why it doesn't work.. @KevinDenys

KevinDenys, 2016-11-25:
Hmmmm I just tried running it on my windows 10 and here it opens the cmd. Looks like my windows 7 is being funky. Thanks for fixing the keyboard layout problem! Issue fixed

closed 2016-11-25 by KevinDenys

Thecakeisgit, 2016-11-25:
You're welcome ! @KevinDenys