Dukweeno/Duckuino #16: MuKeyboard support

MrMoDDoM, 2017-01-23:
Hi,
I create this library called MuKeyboard that enable (with the support from a little circuit) any Arduino/AVR board to support HID interface, thanks to the V-BUS framework.
I ask you if you could include a "flavor" tab for selecting which library use: the default in-build Arduino's HID Keyboard.h or our MuKeyboard.h
There are only few changes for your project:

- Obviously from ```#include "Keyboard.h"``` to ```#include "MuKeyboard.h"```
- Change every ```delay(ms)``` to ```wait(ms)```

Everything else should be perfectly compatible with the Arduino's Keyboard library and with your project indeed.

Here the [link for the repo](https://github.com/MrMoDDoM/MuKeyboard)
Tested on a Arduino Nano and Uno, it works smooth and good!

Cheers,

MrMoDDoM

Thecakeisgit, 2017-01-23:
Hmm why not :D, It's interesting, but we'll have a conflict with [Thecakeisgit/LocaleKeyboard.js](https://github.com/Thecakeisgit/LocaleKeyboard.js) who is a multi locale generator for Keyboard HID Arduino's Lib.
It won't work because your lib isn't the Keyboard.h lib(And this project will generate a custom Keyboard.<cpp|h>).. We'll need to work on this :D

gh7531, 2017-01-23:
Sorry for hijacking this issue but does the MuKeyboard support full 8-bit ASCII or could it be implemented? @MrMoDDoM

MrMoDDoM, 2017-01-23:
@Thecakeisgit I saw that Dckuino uses LocaleKeyboard.js for generating the zip file and I'm thinking how to make much more compatible MuKeyboard with that..
Am I wrong if I say that is only relevant for downloading the zip file? I mean, just for the copy-and-paste guys like me, there should be no problem at this moment..

@gh7531 I'm not sure how to answer your question, but I can say that MuKeyboard, thanks to the V-BUS framework, is able to handle all the key [reported on page 53 of this document](http://www.usb.org/developers/hidpage/Hut1_12v2.pdf)

Thecakeisgit, 2017-01-25:
@MrMoDDoM Not really generating a zip, obviously it's your navigator who generates a zip... LocaleKeyboard.js is generating a custom Keyboard.h & Keyboard.cpp to work with all locales :D 

@gh7531 No problem, but the default library supports already 8-Bit ASCII, I think you mean EASCII (Extended ASCII who isn't supported by the base Keyboard lib)

MrMoDDoM, 2017-01-31:
@Thecakeisgit Sorry, I don't get why LocaleKeyboard could be a problem: because at this moment MuKeyboard is not compatible with LocaleKeyboard.js features, I was thinking that for people who just copy and paste code from Dckuino.js live page should be a good feature to let them choose which "flavor" should your project outputs the results.
I know that this means that the resulting sketch will work only for one locale ( ENG at this moment ).
If you want to wait and have a fully compatible MuKeyboard, I'll work harder on it ;)

@gh7531 I'm still not figured out if MuKeyboard support EASCII (probably yes), but I think it's better to move this question into MuKeyboard's repo.

Thecakeisgit, 2017-02-01:
@MrMoDDoM, Okay if you got the time, you could send me a pull request with you changes :D

MrMoDDoM, 2017-10-12:
Hi, absolutely yes..! 
Could you please leave this issue open just about fews days? 
I'll make it as soon as possible :)

Nurrl, 2017-10-12:
No problem, just duplicate **modules/Arduino.js** in **module/MuKeyboard.js** and add a new line containing `MuKeyboard` in the **modules/modules** file ^^

closed 2017-10-26 by Nurrl