Dukweeno/Duckuino #17: osx wallpaper changer

JonasZaverka, 2017-03-03:
hy i try to translate basic osx wallpaper changer and this popup in my IDE

thx Jonas

Arduino: 1.8.1 (Windows 10), Vývojová deska: "Arduino Leonardo"

osxwall3:1: error: expected constructor, destructor, or type conversion before '(' token

 typeKey(KEY_RETURN);

        ^

osxwall3:3: error: 'Keyboard' does not name a type

   Keyboard.print("cd Desktop/");

   ^

osxwall3:5: error: expected constructor, destructor, or type conversion before '(' token

   typeKey(KEY_RETURN);

          ^

osxwall3:7: error: 'Keyboard' does not name a type

   Keyboard.print("rm 1283877178.or.61533.jpg");

   ^

osxwall3:9: error: expected constructor, destructor, or type conversion before '(' token

   typeKey(KEY_RETURN);

          ^

osxwall3:11: error: expected constructor, destructor, or type conversion before '(' token

   delay(100);

        ^

osxwall3:13: error: 'Keyboard' does not name a type

   Keyboard.print("cd");

   ^

osxwall3:15: error: expected constructor, destructor, or type conversion before '(' token

   typeKey(KEY_RETURN);

          ^

osxwall3:17: error: expected constructor, destructor, or type conversion before '(' token

   delay(50);

        ^

osxwall3:19: error: 'Keyboard' does not name a type

   Keyboard.print("history -c");

   ^

osxwall3:21: error: expected constructor, destructor, or type conversion before '(' token

   typeKey(KEY_RETURN);

          ^

osxwall3:23: error: 'Keyboard' does not name a type

   Keyboard.press(KEY_LEFT_GUI);

   ^

osxwall3:24: error: 'Keyboard' does not name a type

   Keyboard.press('w');

   ^

osxwall3:25: error: 'Keyboard' does not name a type

   Keyboard.releaseAll();

   ^

osxwall3:26: error: expected constructor, destructor, or type conversion before '(' token

   delay(100);

        ^

osxwall3:28: error: 'Keyboard' does not name a type

   Keyboard.press(KEY_LEFT_GUI);

   ^

osxwall3:29: error: 'Keyboard' does not name a type

   Keyboard.press('q');

   ^

osxwall3:30: error: 'Keyboard' does not name a type

   Keyboard.releaseAll();

   ^

osxwall3:33: error: 'Keyboard' does not name a type

   Keyboard.end();

   ^

osxwall3:34: error: expected declaration before '}' token

 }

 ^

exit status 1
expected constructor, destructor, or type conversion before '(' token

Táto zpráva by měla mít víc informacií v
"Zobrazení podrobného výstupu při kompilaci"
podle zapnuté volby v Soubor -> Nastavení.

Plazmaz, 2017-03-03:
You should post the generated sketch file. Looks like you're missing the keyboard library.

chubbybarbarian, 2017-03-17:
you type ENTER not return

Nurrl, 2017-03-18:
Sounds like a syntax error, did you downloaded or copied the script ? 
(Are toi on Linux ? And if yes did you installed Arduino IDE with `apt-get` ?)

Plazmaz, 2017-05-02:
Will close this issue in a bit if we don't get a response from @JonasZaverka

Plazmaz, 2017-05-05:
Closing the issue due to inactivity

closed 2017-05-05 by Plazmaz