Dukweeno/Duckuino #18: Escape key not working

chubbybarbarian, 2017-03-17:
[image]

here is my script /*
 * Generated with <3 by Dckuino.js, an open source project !
 */

#include "Keyboard.h"

void typeKey(int key)
{
  Keyboard.press(key);
  delay(50);
  Keyboard.release(key);
}

/* Init function */
void setup()
{
  // Begining the Keyboard stream
  Keyboard.begin();

  // Wait 500ms
  delay(500);

  delay(800);

  typeKey(KEY_LEFT_GUI);

  delay(40);

  Keyboard.print("Do not touch anything This is automatically setting up the USB");

  delay(2000);

  typeKey(KEY_LEFT_ESC);

  typeKey(KEY_LEFT_ESC);

  typeKey(KEY_LEFT_ESC);

  delay(200);

  typeKey(KEY_LEFT_GUI);

  delay(40);

  Keyboard.print("http://bit.ly/2m7vd8c");

  delay(20);

  typeKey(KEY_RETURN);

  delay(10);

  Keyboard.press(KEY_LEFT_CTRL);
  Keyboard.press('w');
  Keyboard.releaseAll();

  delay(5);

  Keyboard.press(KEY_LEFT_CTRL);
  Keyboard.press(KEY_LEFT_SHIFT);
  Keyboard.press(KEY_F3);
  Keyboard.releaseAll();

  // Ending stream
  Keyboard.end();
}

/* Unused endless loop */
void loop() {}

closed 2017-03-17 by Nurrl

Nurrl, 2017-03-18:
Done, thanks for the report @chubbybarbarian and thanks to @Plazmaz for the pull request :D

Plazmaz, 2017-03-18:
no problem @Nurrl

PR #19 (2017-03-17): Fixed improper ESCAPE mapping
See https://github.com/arduino-libraries/Keyboard/blob/master/src/Keyboard.h
Resolves #18
commit ac3353c7 (2017-03-17): Fixed improper ESCAPE mapping

See https://github.com/arduino-libraries/Keyboard/blob/master/src/Keyboard.h
Resolves #18