Dukweeno/Duckuino #23: AZERTY keyboard...

delopa, 2017-04-01:
I'm french (sorry for my bad english), and I'm using an AZERTY keyboard and I've got some problems.
The code is not correctly executed, So I convert in QUERTY, the same.

Nurrl, 2017-04-02:
Hi @delopa (I'm French too, but for the general understanding I'll talk English ;D),
Did you copied and pasted the code or did you used the download button ? ;D

delopa, 2017-04-02:
Copy / paste, why ?

Nurrl, 2017-04-02:
Yes, this is because as you might know, the code is using the default Arduino Keyboard library(Which only works for English QWERTY layouts), but if you use the **Download Sketch** button, you can select the locale (For you **fr_FR**), and this will bundle a wonderful modified Keyboard library for your locale ;D

Nurrl, 2017-04-02:
Tell me if this doesn't worked, I'll reopen the issue.. ;D

closed 2017-04-02 by Nurrl

Plazmaz, 2017-04-02:
Perhaps disabling copy/paste from the box would be smart when the locale is changed @Nurrl

delopa, 2017-04-03:
It works on Firefox ! Thanks !
But `Start download` didn't work on chrome (why ?)

Nurrl, 2017-04-03:
@Plazmaz Yes it would be cool simply to remake the GUI, and the core to make it modular as I said here : #22.

@delopa I don't know, I'll try to fix it when I'll rebuild the GUI ;D