Dukweeno/Duckuino #24: left arrow and esc

miniminion100, 2017-04-12:
Arduino: 1.8.2 (Windows Store 1.8.2.0) (Windows 10), Board: "Arduino Leonardo"

C:\Users\lsjon\OneDrive\Documents\Arduino\hotdog_backround\hotdog_backround.ino: In function 'void setup()':

hotdog_backround:62: error: 'KEY_LEFT_ESC' was not declared in this scope

   Keyboard.press(KEY_LEFT_ESC);

                  ^

C:\Users\lsjon\OneDrive\Documents\Arduino\hotdog_backround\hotdog_backround.ino: At global scope:

hotdog_backround:114: error: expected initializer at end of input

 void loop() 

           ^

exit status 1
'KEY_LEFT_ESC' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Nurrl, 2017-04-12:
This issue is a duplicate of #20, and has been fixed with ac3353c.

Nurrl, 2017-04-12:
Please recompile the code.

closed 2017-04-12 by Nurrl