Dukweeno/Duckuino #25: This doesn't work how it should work

TJHUNTER2001, 2017-04-13:
I try to open notepad and write some stuff in it but it doesn't work.
Here is the script:

> DELAY 5000
> GUI r
> DELAY 5000
> 
> STRING notepad
> ENTER
> DELAY 5000
> 
> STRING Hello
> DELAY 5000
> ENTER
> DELAY 5000
> ENTER
> DELAY 5000
> STRING Why did you plug this in?

Nurrl, 2017-04-13:
@TJHUNTER2001, this is not a compiler issue while the code compiles..

Are you using an English mapping, and did you used the download button or copy/paste ?
_It doesn't work_ is not really verbose, can you explain ?

TJHUNTER2001, 2017-04-13:
Well, I think I found the problem.
It sends all the keys as a QWERTY keyboard were it should be sent as AZERTY.

For example:
"My name is TJ" becomes "Öz nqöe is TJ"

TJHUNTER2001, 2017-04-13:
Note: it isn't QWERTY, it is QWERTZ

TJHUNTER2001, 2017-04-13:
One solution I found is to press ALT+SHIFT and then write the strings.

Then the code would look like this:

> DELAY 5000
> GUI r
> DELAY 5000
> 
> STRING notepqd
> DELAY 5000
> ENTER
> DELAY 5000
> 
> STRING Hello
> DELAY 5000
> ENTER
> DELAY 5000
> ENTER
> DELAY 5000
> STRING Yhz did zou plug this inM

TJHUNTER2001, 2017-04-13:
This is another solution I came up with:

> DELAY 5000
> GUI r
> DELAY 5000
> 
> STRING notepqd
> DELAY 100
> ENTER
> DELAY 2000
> 
> ALT SHIFT
> ALT SHIFT
> DELAY 100
> STRING Hello
> DELAY 100
> ENTER
> DELAY 50
> ENTER
> DELAY 50
> STRING Why did you plug this in?
> ALT SHIFT

TJHUNTER2001, 2017-04-15:
I still need some help with this, could anyone give a solution or tell me were to ask pls.

Plazmaz, 2017-04-15:
Please provide the following information to help us address the problem more easily:
The original duckyscript code
The Arduino output or the generated Arduino project
Your keyboard layout/language
The full output of the script
The expected output

TJHUNTER2001, 2017-04-16:
- [x ] The original duckyscript code : Already posted
- [x ] The Arduino output or the generated Arduino project : See video
- [x ] Your keyboard layout/language : AZERTY (Dutch if it mathers)
- [x ] The full output of the script : See video
- [x ] The expected output : See video

PS: videos'll come later

TJHUNTER2001, 2017-04-16:
Output original code:

> /*
>  * Generated with <3 by Dckuino.js, an open source project !
>  */
> 
> #include "Keyboard.h"
> 
> void typeKey(int key)
> {
>   Keyboard.press(key);
>   delay(50);
>   Keyboard.release(key);
> }
> 
> /* Init function */
> void setup()
> {
>   // Begining the Keyboard stream
>   Keyboard.begin();
> 
>   // Wait 500ms
>   delay(500);
> 
>   delay(5000);
> 
>   Keyboard.press(KEY_LEFT_GUI);
>   Keyboard.press('r');
>   Keyboard.releaseAll();
> 
>   delay(5000);
> 
>   Keyboard.print("notepad");
> 
>   typeKey(KEY_RETURN);
> 
>   delay(5000);
> 
>   Keyboard.print("Hello");
> 
>   delay(5000);
> 
>   typeKey(KEY_RETURN);
> 
>   delay(5000);
> 
>   typeKey(KEY_RETURN);
> 
>   delay(5000);
> 
>   Keyboard.print("Why did you plug this in?");
> 
>   // Ending stream
>   Keyboard.end();
> }
> 
> /* Unused endless loop */
> void loop() {}

TJHUNTER2001, 2017-04-16:
Download file (.avi format) at:
https://drive.google.com/open?id=0By4Bw1LyCjW1MzM0MFpoVTdJanc

Plazmaz, 2017-04-16:
Try using the "Download Sketch" function instead of copying+pasting the code.

TJHUNTER2001, 2017-04-16:
Same result :(

Nurrl, 2017-04-16:
By selecting the good locale.... @TJHUNTER2001

TJHUNTER2001, 2017-04-16:
What do you mean with good locale?

Nurrl, 2017-04-18:
I mean an AZERTY locale like **fr_FR** :D

TJHUNTER2001, 2017-04-18:
Sorry, feel me like a noob but how do I do that?

Nurrl, 2017-04-18:
Click the **Download Sketch** button, then select the locale and click **Download** :D

TJHUNTER2001, 2017-04-18:
Didn't help :'(

Andsoeg, 2017-04-24:
This line;  Keyboard.print("cd \"%USERPROFILE%\\Desktop\" & for /f \"tokens=2 delims=:\" %A in ('netsh wlan show interface ^| findstr \"SSID\" ^| findstr /v \"BSSID\"') do set A=%A");

Turned out like this in console;

cd Æ%USERPROFILE%'DesktopÆ / for -f Ætokens\2 delims\ØÆ %A in )ænetsh wlan show interface &* findstr ÆSSIDÆ &* findstr -v ÆBSSIDÆæ= do set A\%A

Does the rubber ducky (arduino) use the system language keyboard? Very strange. If i plug a regular keyboard into my pc, that keyboard uses a standalone language.

Nurrl, 2017-04-24:
@Andsoeg On Windows you could add an <code>ALT SPACE</code> before the code...

TJHUNTER2001, 2017-05-12:
Alt space? Why?

Nurrl, 2017-05-12:
No, forget this, ALT SPACE is to swap locales on Windows, but you need to have the english one activated .-. Worst idea I ever talked about x)

TJHUNTER2001, 2017-06-08:
Windows lang (unknown) on testing sub it's Dutch.

exzotic, 2017-07-17:
help i have done these steps used the download and tried the french locale on the downloader still having problems

Nurrl, 2017-07-17:
Hi @exzotic, what did you done ?

Nurrl, 2017-10-11:
Hi @TJHUNTER2001, @Andsoeg and @exzotic, the **v2.0-pre** has been released, can you try again to compile the code and send it to your Arduino ?

closed 2018-04-28 by Nurrl