Dukweeno/Duckuino #26: Lay out goals and thoughts for v2.0

Plazmaz, 2017-04-18:
I suggest laying out your thoughts and opening discussion for the v2.0 you've mentioned via an issue, and creating a milestone with the results.

Nurrl, 2017-05-02:
Yep, (I'm sorry, I really didn't saw this !), I managed to get a working Dckuino.js version with modules, I also recreated the GUI, so it's more user-friendly :D I need to fix some bugs and issues, but I think I'll release it soon ;D
Also, I'm very unfamiliar with Milestone, would you like to be added as contributor, so you can create it ? @Plazmaz

Plazmaz, 2017-05-02:
I mentioned it in a comment on the last issue, but is there some way we could get in contact via email or private message of some sort?

Plazmaz, 2017-05-02:
Your previous email doesn't seem to work

Nurrl, 2017-05-02:
Oh, whoops, I changed it :3 Go hack.chat ?

Nurrl, 2017-05-02:
https://hack.chat/?dckuino

closed 2017-06-12 by Plazmaz