Dukweeno/Duckuino #27: Payload does not work

pigoof, 2017-05-02:
Hi there,

I want to convert this ducky script to be readable for my arduino to execute but it does not convert because something about line 12 it says it doesnt understand A

Ill send you the ducky script if you can convert it for my - id appreciate it alot thanks!

DELAY 5000
SPACE
DELAY 500
SPACE
DELAY 500
DOWN
DELAY 500
ENTER
DELAY 2000
STRING ************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************** [... cut ...]
DELAY 500
CTRL V
DELAY 1000
CTRL A
DELAY 500
CTRL C
DELAY 500
CTRL V
DELAY 1000
CTRL A
DELAY 500
CTRL C
DELAY 500
CTRL V
DELAY 1000
CTRL A
DELAY 500
CTRL C
DELAY 500
CTRL V
DELAY 1000
CTRL A
DELAY 500
CTRL X
DELAY 500
ESC
DELAY 15000
CTRL V
DELAY 1000
LEFT
LEFT
LEFT
LEFT
LEFT
DELAY 500
RIGHT
RIGHT
RIGHT
[... 107 lines cut ...]
LEFT
DELAY 500
RIGHT
RIGHT
RIGHT
RIGHT
RIGHT
DELAY 1000
CTRL V
DELAY 1000
LEFT
LEFT
LEFT
LEFT
LEFT
DELAY 500
RIGHT
RIGHT
RIGHT
RIGHT
RIGHT
DELAY 1000
CTRL V
DELAY 500
STRING 11111

Plazmaz, 2017-05-02:
The keys(CTRL + A, CTRL + C, etc) should be lowercase. Meaning your code should be this:

```
CTRL s
DELAY 500
CTRL c
DELAY 500
CTRL v
DELAY 1000
CTRL a
DELAY 500
CTRL c
DELAY 500
CTRL v
DELAY 500
CTRL a
DELAY 500
CTRL c
DELAY 500
CTRL v
DELAY 1000
CTRL a
DELAY 500
CTRL c
DELAY 500
CTRL v
DELAY 1000
CTRL a
DELAY 500
CTRL c
DELAY 500
CTRL v
DELAY 1000
CTRL a
DELAY 500
CTRL c
DELAY 500
CTRL v
DELAY 1000
CTRL a
DELAY 500
CTRL x
DELAY 500
ESC
DELAY 15000
CTRL v
DELAY 1000
LEFT
LEFT
LEFT
LEFT
LEFT
DELAY 500
RIGHT
RIGHT
RIGHT
RIGHT
RIGHT
DELAY 1000
CTRL v
DELAY 1000
LEFT
LEFT
LEFT
LEFT
LEFT
DELAY 500
RIGHT
RIGHT
RIGHT
RIGHT
RIGHT
DELAY 1000
CTRL v
DELAY 1000
LEFT
LEFT
LEFT
LEFT
LEFT
DELAY 500
RIGHT
RIGHT
RIGHT
RIGHT
RIGHT
DELAY 1000
CTRL v
DELAY 1000
LEFT
LEFT
LEFT
LEFT
LEFT
DELAY 500
RIGHT
RIGHT
RIGHT
RIGHT
RIGHT
DELAY 1000
CTRL v
DELAY 1000
LEFT
LEFT
LEFT
LEFT
LEFT
DELAY 500
RIGHT
RIGHT
RIGHT
RIGHT
RIGHT
DELAY 1000
CTRL v
DELAY 1000
LEFT
LEFT
LEFT
LEFT
LEFT
DELAY 500
RIGHT
RIGHT
RIGHT
RIGHT
RIGHT
DELAY 1000
CTRL v
DELAY 1000
LEFT
LEFT
LEFT
LEFT
LEFT
DELAY 500
RIGHT
RIGHT
RIGHT
RIGHT
RIGHT
DELAY 1000
CTRL v
DELAY 1000
LEFT
LEFT
LEFT
LEFT
LEFT
DELAY 500
RIGHT
RIGHT
RIGHT
RIGHT
RIGHT
DELAY 1000
CTRL v
DELAY 1000
LEFT
LEFT
LEFT
LEFT
LEFT
DELAY 500
RIGHT
RIGHT
RIGHT
RIGHT
RIGHT
DELAY 1000
CTRL v
DELAY 1000
LEFT
LEFT
LEFT
LEFT
LEFT
DELAY 500
RIGHT
RIGHT
RIGHT
RIGHT
RIGHT
DELAY 1000
CTRL v
DELAY 500
STRING 11111
```

pigoof, 2017-05-02:
It partially worked but the code came across a problem... 

Arduino: 1.8.2 (Windows 7), Board: "Arduino Leonardo"

Duckduino-microSD:165: error: 'KEY_LEFT_ESC' was not declared in this scope

??? know what went wrong?

pigoof, 2017-05-02:
I appreciate your help.
I dont know. latest version of what? 

I got it to work by changing the error key_left_esc to key_left_CTRL - not sure if thats good or bad...
The only problem after that is there isnt enough room on the arduino.

Im following this guys instructions to put the code onto sd card -https://www.youtube.com/watch?v=ksvo1WDYQ7s

But it doesnt work... only when the arduino has something on it does it work - the sd card is as if it wasnt connected (because it doesnt execute anything.)

I must be missing something..?

Plazmaz, 2017-05-02:
The space issue can't be helped. I suggest using for loops in the generated code. Unfortunately, the channel you linked suggests the use of an outdated version of duckuino(hosted on seytonic.com). The bug you mentioned has been fixed previously. I suggest using the latest version we host on GitHub, which you can find if you look at the readme.md.

pigoof, 2017-05-02:
Okay so I used the converter like you said but how do I get it to work from sd card or put it on arduino from your converter?

This is the new code: 
/* Converted by Duckuino:
* https://forums.hak5.org/index.php?/topic/32719-payload-converter-duckuino-duckyscript-to-arduino/?p=244590
* Enjoy!
*/
void setup() {
	Keyboard.begin();
    
type(KEY_LEFT_CTRL,false);
type('S',false);
Keyboard.releaseAll();
delay(500);
    
type(KEY_LEFT_CTRL,false);
type('C',false);
Keyboard.releaseAll();
delay(500);
    
type(KEY_LEFT_CTRL,false);
type('V',false);
Keyboard.releaseAll();
delay(1000);
    
type(KEY_LEFT_CTRL,false);
type('A',false);
Keyboard.releaseAll();
delay(500);
    
type(KEY_LEFT_CTRL,false);
type('C',false);
Keyboard.releaseAll();
delay(500);
    
type(KEY_LEFT_CTRL,false);
type('V',false);
Keyboard.releaseAll();
delay(500);
    
type(KEY_LEFT_CTRL,false);
type('A',false);
Keyboard.releaseAll();
delay(500);
    
type(KEY_LEFT_CTRL,false);
type('C',false);
Keyboard.releaseAll();
delay(500);
    
type(KEY_LEFT_CTRL,false);
type('V',false);
Keyboard.releaseAll();
delay(1000);
    
type(KEY_LEFT_CTRL,false);
type('A',false);
Keyboard.releaseAll();
delay(500);
    
type(KEY_LEFT_CTRL,false);
type('C',false);
Keyboard.releaseAll();
delay(500);
    
type(KEY_LEFT_CTRL,false);
type('V',false);
Keyboard.releaseAll();
delay(1000);
    
type(KEY_LEFT_CTRL,false);
type('A',false);
Keyboard.releaseAll();
delay(500);
    
type(KEY_LEFT_CTRL,false);
type('C',false);
Keyboard.releaseAll();
delay(500);
    
type(KEY_LEFT_CTRL,false);
type('V',false);
Keyboard.releaseAll();
delay(1000);
    
type(KEY_LEFT_CTRL,false);
type('A',false);
Keyboard.releaseAll();
delay(500);
    
type(KEY_LEFT_CTRL,false);
type('C',false);
Keyboard.releaseAll();
delay(500);
    
type(KEY_LEFT_CTRL,false);
type('V',false);
Keyboard.releaseAll();
delay(1000);
    
type(KEY_LEFT_CTRL,false);
type('A',false);
Keyboard.releaseAll();
delay(500);
    
type(KEY_LEFT_CTRL,false);
type('X',false);
Keyboard.releaseAll();
delay(500);
    
type(KEY_LEFT_ESC,false);
Keyboard.releaseAll();
delay(15000);
    
type(KEY_LEFT_CTRL,false);
type('V',false);
Keyboard.releaseAll();
delay(1000);
    for(int i = 0; i < 5; i++) {
    LEFT
    }
delay(500);
    RIGHT
    RIGHT
    for(int i = 0; i < 5; i++) {
    RIGHT
    }
delay(1000);
    LEFT
    LEFT
    LEFT
    LEFT
    for(int i = 0; i < 5; i++) {
    LEFT
    }
    RIGHT
    RIGHT
delay(1000);
    for(int i = 0; i < 5; i++) {
    RIGHT
    }
    LEFT
    LEFT
delay(500);
    RIGHT
    RIGHT
    for(int i = 0; i < 5; i++) {
    LEFT
    }
delay(1000);
    LEFT
    LEFT
    for(int i = 0; i < 5; i++) {
    RIGHT
    }
    RIGHT
    RIGHT
    RIGHT
    RIGHT
delay(1000);
    for(int i = 0; i < 5; i++) {
    LEFT
    }
    LEFT
    LEFT
delay(500);
    for(int i = 0; i < 5; i++) {
    RIGHT
    }
delay(1000);
    
type(KEY_LEFT_CTRL,false);
type('V',false);
Keyboard.releaseAll();
delay(1000);
    LEFT
    LEFT
    for(int i = 0; i < 5; i++) {
    LEFT
    }
    RIGHT
    RIGHT
    RIGHT
    for(int i = 0; i < 5; i++) {
    RIGHT
    }
    LEFT
    LEFT
    LEFT
    LEFT
delay(500);
    for(int i = 0; i < 5; i++) {
    LEFT
    }
delay(1000);
    
type(KEY_LEFT_CTRL,false);
type('V',false);
Keyboard.releaseAll();
delay(1000);
    for(int i = 0; i < 5; i++) {
    RIGHT
    }
delay(500);
    RIGHT
    RIGHT
    RIGHT
    RIGHT
    for(int i = 0; i < 5; i++) {
    LEFT
    }
    LEFT
    LEFT
    LEFT
    for(int i = 0; i < 5; i++) {
    RIGHT
    }
    RIGHT
    RIGHT
delay(1000);
    
type(KEY_LEFT_CTRL,false);
type('V',false);
Keyboard.releaseAll();
delay(1000);
    for(int i = 0; i < 5; i++) {
    LEFT
    }
delay(500);
    RIGHT
    RIGHT
    for(int i = 0; i < 5; i++) {
    RIGHT
    }
delay(500);

print(F("11111"));
    for(int i = 0; i < 5; i++) {
    LEFT
    }
    for(int i = 0; i < 5; i++) {
    RIGHT
    }
    for(int i = 0; i < 5; i++) {
    LEFT
    }
    for(int i = 0; i < 5; i++) {
    RIGHT
    }
Keyboard.end();
}
void type(int key, boolean release) {
	Keyboard.press(key);
	if(release)
		Keyboard.release(key);
}
void print(const __FlashStringHelper *value) {
	Keyboard.print(value);
}
void loop(){}

Plazmaz, 2017-05-02:
Sorry, should've been more specific, use this page https://nurrl.github.io/Dckuino.js/. As for the SD card, that is unrelated to this project, I suggest asking the creator of that youtube video. I think this issue is resolved now. Care to close it, @Nurrl?

Nurrl, 2017-05-02:
I See ! He use you old Duckuino @Plazmaz, so please use the now live version ! Link above :D 
Let us know if it works ^^

Nurrl, 2017-05-02:
Yep no problem, I'll mark it as invalid while this is referred to the old project :D @Plazmaz

closed 2017-05-02 by Nurrl

Plazmaz, 2017-05-02:
Just FYI, I chatted with the author of that video, and he said he would update to the latest version. On another note, can I contact you via email or some other platform @Nurrl?

pigoof, 2017-05-03:
Thanks alot , I'll give it a go &see how it works. I'll let you know of my updates. If you want you can contact me by this email - mytonicstore@hotmail.com

pigoof, 2017-05-03:
Your link doesnt work - error 404 - I cant open link. https://nurrl.github.io/Dckuino.js/.

Plazmaz, 2017-05-03:
@pigoof thanks! the link is https://nurrl.github.io/Duckuino/

pigoof, 2017-05-03:
Great! 
Okay so all is working the only problem is the size is too big. Now I just started in programming a few days ago so this is all pretty new to me... Would it be possible to shorten the file? If so how?

Plazmaz, 2017-05-03:
I suggest using for loops and functions to reduce the size. I'll likely be implementing some optimization systems soon to reduce filesize drastically. Another thing you could do is remove anything you don't need. For instance, if there's a long sequence of text, see if you can shorten it. Other than that, there's not much I can do for you, sorry.

pigoof, 2017-05-04:
What do you mean ? Loops and functions? It would be great if you could explain what those are or do it to the code for me. Anyways I appreciate you sticking with me and providing fast replys. I wasn't expecting such commitment.  :) Thanks!!!

Plazmaz, 2017-05-04:
@pigoof No problem, however it is not my job to teach programming concepts, and I'm not going to write it for you, sorry. I would suggest looking into some basic C++ concepts. If not, maybe try looking at https://github.com/hak5darren/USB-Rubber-Ducky/wiki/Duckyscript#replay for loops.

pigoof, 2017-05-04:
Perfectly understandable! :)   I have a lot of research ahead of me, appreciate your time with me!