Dukweeno/Duckuino #28: ram optimising

Flambello, 2017-05-06:
hello
instead of Keyboard.print("hello world");
 can u make it do Keyboard.print(F("hello world));
so that the print will not get stored in the ram and instead go in the flash memory making longer line scripts possible

the foto is of a verry long script 
without the (F()) it uses around 8k ram
with the (F()) it uses only 220 

[image]

Plazmaz, 2017-05-07:
This is something I used in the original version of Duckuino, but it seems to have gotten lost in translation. However, writing everything to flash may not be the best way of doing things, otherwise, it would probably be the default, I'll need to look into this more

Flambello, 2017-05-07:
Oooh ok thnx for the responce

Nurrl, 2017-05-07:
@Plazmaz You could add it in your post-processor :D Or we could add a checkbox to let the user decide ^^

Flambello, 2017-05-07:
Uuuu that whould be nice!

Plazmaz, 2017-05-07:
I think that sounds good. I like the idea of being able to disable it.

Flambello, 2017-05-07:
Haha best software on github!

Plazmaz, 2017-05-07:
@Flambello if you think it's good now, get ready for v2.

Flambello, 2017-05-08:
No problem im glad my suggestion helps!!!

bytedeez, 2017-06-15:
@Flambello @Plazmaz 
I have about 5 scripts i need to use this for. I am still learning .js and jquery at the moment
Where in the code could i make this change temporarily? 
I searched through the code but nothing stuck out to a novice like myself to make this change without some expert advice,
Thanks in advance!

Plazmaz, 2017-06-15:
@bytedeez consider it done.

Nurrl, 2017-10-11:
Since v2 is here and this disappeared, I'm reopening this, @Plazmaz should release the Ram Optimization update some days :D

Plazmaz, 2017-10-11:
@Nurrl nope, this is something different. This issue is asking for the use of Fstrings instead of normal strings. Should be an easy fix 😄

Nurrl, 2017-10-11:
Yep ^^ But it Optimizes ram x)

closed 2017-10-11 by Nurrl