Dukweeno/Duckuino #29: No DEFAULT_DELAY

keeganjk, 2017-05-22:
When you try to use `DEFAULT_DELAY` or `DEFAULTDELAY` to set a default delay between each command, it results in the error, `An error occured, the compiler returned undefined content`.

Nurrl, 2017-05-23:
Hi, @keeganjk thanks for reporting this error, but we need more informations please :)
Please provide us : 
  - The uncompiled script
  - What the compiler says (On the bottom of the page, the long bar ;D), this should help us to solve your problem..

Plazmaz, 2017-05-23:
@Nurrl I just think we haven't implemented this command. I remember noticing this at one point.

Nurrl, 2017-05-23:
@Plazmaz [Dckuino.js Line 233](https://github.com/Nurrl/Duckuino/blob/master/Dckuino.js#L233) I implemented it when I added the REPEAT command :)

keeganjk, 2017-06-03:
I've updated my comment, I've added some info.

Nurrl, 2017-06-04:
I know, but when I use it, it works... @keeganjk

Nurrl, 2017-06-04:
[image]

keeganjk, 2017-06-04:
@Nurrl Not for me... 
[image]

Plazmaz, 2017-06-04:
@keeganjk please ensure you are running the code yourself, versus using someone else's hosted version. We've had issues with version conflicts in hosted versions of the project

keeganjk, 2017-06-05:
@Plazmaz I just downloaded it to see if the latest version worked. It didn't work.

Plazmaz, 2017-06-05:
@keeganjk I am able to reproduce with just one line, but when there's other data in the editor, it does not occur

keeganjk, 2017-06-05:
@Plazmaz @Nurrl I've found the problem! It requires multiple lines and some hosted versions are outdated.

closed 2017-06-05 by keeganjk

Plazmaz, 2017-06-05:
@Nurrl perhaps we should give a better error for this.