Dukweeno/Duckuino #30: Language error

loidor, 2017-07-04:
When I'm plugging in my arduino on an English computer everything works flawlessly, but when plugging it in on a Swedish one everything ends up wonky. / is for example replaced by -. 

This is due to keyboard layout (Swedish keyboards does indeed have - where English ones have /). Would it be possible to force the script to adapt to the current keyboard layout?

Nurrl, 2017-07-04:
Hi @loidor, this project includes **LocaleKeyboard.js**, which is an other project, and fixes this bug, but there is currently no Swedish locale.

loidor, 2017-07-04:
Lovely, then there's a way to solve it :) I'll look into generating a Swedish contribution to **LocaleKeyboard.js**. I close this since it's kind of solved.

closed 2017-07-04 by loidor