Dukweeno/Duckuino #33: Error When Writing to a Arduino Leo!!!

TheEnchantedCow, 2017-09-07:
Arduino: 1.8.4 Hourly Build 2017/08/18 04:33 (Windows 10), Board: "Arduino Leonardo"

C:\Users\Adminstrator\Documents\Arduino Rubber Ducky\Test\Dckuino.js-yd1w.ino: In function 'void typeKey(uint8_t)':

Dckuino.js-yd1w:7: error: redefinition of 'void typeKey(uint8_t)'

 void typeKey(uint8_t key)

      ^

C:\Users\Adminstrator\Documents\Arduino Rubber Ducky\Test\Test.ino:7:6: note: 'void typeKey(uint8_t)' previously defined here

 void typeKey(uint8_t key)

      ^

C:\Users\Adminstrator\Documents\Arduino Rubber Ducky\Test\Dckuino.js-yd1w.ino: In function 'void setup()':

Dckuino.js-yd1w:15: error: redefinition of 'void setup()'

 void setup()

      ^

C:\Users\Adminstrator\Documents\Arduino Rubber Ducky\Test\Test.ino:15:6: note: 'void setup()' previously defined here

 void setup()

      ^

C:\Users\Adminstrator\Documents\Arduino Rubber Ducky\Test\Dckuino.js-yd1w.ino: In function 'void loop()':

Dckuino.js-yd1w:48: error: redefinition of 'void loop()'

 void loop() {}

      ^

C:\Users\Adminstrator\Documents\Arduino Rubber Ducky\Test\Test.ino:48:6: note: 'void loop()' previously defined here

 void loop() {}

      ^

exit status 1
redefinition of 'void typeKey(uint8_t)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Plazmaz, 2017-09-07:
Looks like the script is included twice. Can you send me your code?
Also, why do you have `Test.ino` and `Dckuino.js-yd1w.ino`?

TheEnchantedCow, 2017-09-07:
Yeah... Thanks for the speedy response by the way!! 

/*
 * Generated with <3 by Dckuino.js, an open source project !
 */

#include "Keyboard.h"

void typeKey(uint8_t key)
{
  Keyboard.press(key);
  delay(50);
  Keyboard.release(key);
}

/* Init function */
void setup()
{
  // Begining the Keyboard stream
  Keyboard.begin();

  // Wait 500ms
  delay(500);

  delay(3000);

  Keyboard.press(KEY_LEFT_GUI);
  Keyboard.press('r');
  Keyboard.releaseAll();

  delay(500);

  Keyboard.print(F("notepad"));

  delay(500);

  typeKey(KEY_RETURN);

  delay(750);

  Keyboard.print(F("Hello World!!!"));

  typeKey(KEY_RETURN);

  // Ending stream
  Keyboard.end();
}

/* Unused endless loop */
void loop() {}

Plazmaz, 2017-09-08:
No problem! 
Okay, so it looks like you have two sketches within the same directory. I'm pretty sure this isn't allowed by Arduino by default. Try moving either `Test.ino` or ` Dckuino.js-yd1w.ino` to another directory

TheEnchantedCow, 2017-09-14:
Yeah It was that. I got it to work. This is amazing!

Plazmaz, 2017-09-14:
Happy to hear it @TheEnchantedCow! Will be closing this issue now.

closed 2017-09-14 by Plazmaz