Dukweeno/Duckuino #36: problems

FCucumberGames, 2017-10-01:
[image]
How does it works??

Nurrl, 2017-10-03:
Maybe try to set your Board to **Arduino Mini** ?

FCucumberGames, 2017-10-03:
Thanks, I would try

FCucumberGames, 2017-10-03:
[image]
OOPS, it doesn't work

Nurrl, 2017-10-03:
Have you updated to the latest Arduino IDE version ? @FCucumberGames

FCucumberGames, 2017-10-03:
Shit, I'm fucking retard. I use Arduino Nano

Nurrl, 2017-10-03:
**Arduino Nano** should work tho..

FCucumberGames, 2017-10-03:
Hmm, so, I don't know why I cannot complie this code

FCucumberGames, 2017-10-03:
It's 1.8.4 arduino IDE version

Nurrl, 2017-10-03:
There is the 1.8.5 update, but I don't thinks that's the problem..
I also noticed that the first line chnaged betwin your two screenshots: 
`#include <Keyboard.h>` to `#include ""Keyboard.h"`, maybe that could be the problem..

FCucumberGames, 2017-10-03:
I have tried all variants of this line like "Keyboard.h", <Keyboard.h>, 'Keyboard.h'

FCucumberGames, 2017-10-03:
Thank you for help, but I have burned my arduino while I was brazing a button to it(((

Plazmaz, 2017-10-03:
@FCucumberGames  D: Oh noes!
https://github.com/Nurrl/Duckuino/issues/12#issuecomment-261359817 temporary workaround(too late sadly)

Plazmaz, 2017-10-03:
@Nurrl we should find a more permanent fix. Also, go ahead and push v2 if you think it's ready. I'll worry about the code optimizer later.

Nurrl, 2017-10-03:
@Plazmaz The problem is that v2 isn't ready, I didn't made the support for DEFAULTDELAY..

Nurrl, 2017-10-03:
And we need to figure out how to load modules without *evil*(**eval()**) also..

Nurrl, 2017-10-11:
Hi @FCucumberGames, the **v2.0-pre** has been released, can you try to compile your code again ?

FCucumberGames, 2017-10-13:
No, it still doesn't work. It's the same problem

Nurrl, 2018-08-19:
@FCucumberGames If you still have the problem, I'll reopen this.

closed 2018-08-19 by Nurrl