Dukweeno/Duckuino #45: Issue I found when using GUI r

ITHzWANip, 2017-10-28:
*Short introduction to your problem.*

__Input code:__
```vb
DELAY 3000
GUI r
DELAY 500
STRING notepad
DELAY 500
ENTER
DELAY 750
STRING Hello World!!!
ENTER
```

__Output code:__
```arduino
/**
 * Made with Duckuino, an open-source project.
 * Check the license at 'https://github.com/Nurrl/Duckuino/blob/master/LICENSE'
 */

#include "Keyboard.h"

void typeKey(uint8_t key)
{
  Keyboard.press(key);
  delay(50);
  Keyboard.release(key);
}

/* Init function */
void setup()
{
  // Begining the Keyboard stream
  Keyboard.begin();

  // Wait 500ms
  delay(500);

  delay(3000);
  Keyboard.press(KEY_LEFT_GUI);
  Keyboard.press("r");
  Keyboard.releaseAll();

  delay(500);
  Keyboard.print(F("notepad"));

  delay(500);
  typeKey(KEY_RETURN);

  delay(750);
  Keyboard.print(F("Hello World!!!"));

  typeKey(KEY_RETURN);

  // Ending stream
  Keyboard.end();
}

/* Unused endless loop */
void loop() {}
```

__Error:__
```js
the issue is with the Keyboard.press("r"); 
the r should be in single quotations like Keyboard.press('r')

Using the double quotations caused every icon on my desktop
to continuously open untill I unplugged the arduino
```

Nurrl, 2017-10-28:
Hi @ITHzWANip, are you sure that when you change `"r"` to `'r'`, the bug doesn't happen ?
What you're explaining make me think more about an Arduino power issue, like rebooting continuously and spamming `<enter>`...

ITHzWANip, 2017-10-28:
yea i would screen cap it but for whatever reason when i have my recording software up it wont replicate. But I've tested it multiple times and 'r' works over "r" every time. My theory was that any press needs to be in single quotes but that was not reliable in my testing. It might just be the board im using. Its the pro micro.

nixublx, 2017-10-29:
I noticed the double quotes are passed in the keyArray parameter of the module's computeKeys callback. For the Digispark module I already needed to strip these double quotes, as characters need to be converted to constants ( 'a' becomes KEY_A, etc.) Would it be an idea to omit the double quotes and have the board module decide how to format the character?

Nurrl, 2017-10-29:
@nixublx Yep I'll do that ^^

Nurrl, 2017-10-30:
For the v2.1, and I'll make per module locale selection by merging LocaleKeyboard.js in Duckuino ^^
@Plazmaz Do you agree with these changes ?

Plazmaz, 2017-10-30:
@Nurrl Yep, sounds good

Nurrl, 2017-10-30:
@Plazmaz Great ^^