Dukweeno/Duckuino #50: How to set set decoder to another language ?

akaped, 2018-01-09:
Hello,
I would like to know if it has been implemented the possibility to set another keyboard language different than English in the decoder. I'm Italian and obviously the code I translate with this software types keys that on my Italian keyboard don't make much sense. 

The "original" version of the Rubbery Ducky the script compiler it's also able to compile the code with a language option, to adapt it to different keyboards. 
Can this be done with this software ? If not,  are you planning an update that will enable this mapping function? 

Thanks ! Amazing software BTW 👍

Plazmaz, 2018-01-18:
@Nurrl I believe you have something for different layouts, right?

Nurrl, 2018-01-18:
Yes, @akaped, Normally language compiling is fully integrated with Duckuino, you select your language before downloading.. 
Maybe I don't get your issue..

closed 2018-04-28 by Nurrl