Dukweeno/Duckuino #58: Help? This is weird..

acujl, 2018-02-19:
The Problem basicly: http://prntscr.com/ih1o8d

If u don't trust the link..

**Code:** 
#include "Keyboard.h"

void typeKey(int key)
{
  Keyboard.press(key);
  delay(50);
  Keyboard.release(key);
}

void setup()
{
  // Begining the Keyboard stream
  Keyboard.begin();

  // Wait 500ms
  delay(500);

  delay(3000);

  Keyboard.press(KEY_LEFT_GUI);
  Keyboard.press('r');
  Keyboard.releaseAll();

  delay(500);

  Keyboard.print("notepad");

  delay(500);

  typeKey(KEY_RETURN);

  delay(750);

  Keyboard.print("Hello World!!!");

  typeKey(KEY_RETURN);

  // Ending stream
  Keyboard.end();
}

/* Unused endless loop */
void loop() {}

**Problem:**

exit status 1
'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?

Nurrl, 2018-02-20:
Hi @acujl, to me it looks like you're trying to upload this code on an unsuported board..
On what type of Arduino(or Clone) are you trying to make this work with ?

acujl, 2018-02-20:
Yes, (well probably a clone yeah) it's a Pro Micro. With a processor Atmel MEGA32U4 as requested to "transform" the arduino into a bad usb (?)

Well I already have the CH341SER driver installed, I don't know what's wrong.. But I guess that the computer doesn't even detects it
[image]

Nurrl, 2018-02-20:
It should be good to go, but in the tools tab of the Arduino IDE, please make sure that "Arduino Pro Micro" is selected, then you'll be able to compile the Sketch !

Nurrl, 2018-02-22:
Did it worked ? @acujl

acujl, 2018-02-22:
My PC isn't able to detect it for some reason :/ any thoughts on that? Or do you know other way to program it? I have uno too!

acujl, 2018-02-23:
Well problem solved, just changed the cable :V 
Sorry for the inconvenience! 

P.S: Showed up as Leonardo but worked just fine!

Nurrl, 2018-02-23:
No problem @acujl, most of the Chinese cables are charge-only, I did thought about that :v

closed 2018-02-23 by Nurrl