Dukweeno/Duckuino #59: A small problem

Wesz808, 2018-02-23:
With every script I convert I get the same problem. 
The first thing that is executed on Windows is the magnifier. And I have seriously no idea why. 
The rest of the code executes without any problems. 

So when I convert for example this Ducky Script

> DELAY 3000
> GUI r
> DELAY 600
> STRING cmd
> DELAY 200
> ENTER
> DELAY 750
> STRING shutdown -s -t 5
> ENTER

The result is:
It opens magnifier
and prints:

> cmd
> shutdown -s -t 5

In whatever screen is open. 

So my first thought was that it is a keyboard issue. But I got a standard QWERTY keyboard set on English. This is probably a complete noobish question, but I would really appreciate the help.

Nurrl, 2018-02-23:
Hi @Wesz808, simple question, but on what hardware are you running this code ?

Wesz808, 2018-02-23:
It's running it on a Arduino Leonardo USB. The only thing my own research came to was it has something to do with HID functionality? Since i'm new to this, I really miss some basics which i'm trying to learn. 
It gives this error as output I see now.  

> C:\----\Sketch\Sketch.ino: In function 'void setup()':
> 
> C:\----\Sketch\Sketch.ino:26:21: warning: invalid conversion from 'const char*' to 'uint8_t {aka unsigned char}' [-fpermissive]
> 
>    Keyboard.press("r");
> 
>                      ^
> 
> In file included from C:\----\Sketch\Sketch.ino:6:0:
> 
> sketch\Keyboard.h:92:10: note: initializing argument 1 of 'size_t Keyboard_::press(uint8_t)'
> 
>    size_t press(uint8_t k);
> 
>           ^
>

Nurrl, 2018-02-23:
These are only warnings, since the Sketch compiles, I think this has nothing to do with the code..
What are your thought @Plazmaz ?

Plazmaz, 2018-02-23:
Hmm. It doesn't look to be a compilation error. Try increasing your delays to run very slowly.

Wesz808, 2018-02-24:
Already tried. Also, a different USB. Got the same result (only slower). I'll try compiling it using a different computer. Don't think it will make a difference. But can't hurt excluding possibilities...

Wesz808, 2018-02-24:
Same problem on my laptop (was using my desktop before). Same error, it compiles but opens the magnifier is executed.

Nurrl, 2018-03-07:
Hmm, that's very annoying, the magnifier shortcut is `GUI` + `+`. So the Arduino prints `+` instead of `r` ? Can you try to compile and use the following DuckyScript code :
```vbs
TYPE r
```
And execute it with notepad opened to see if it prints a `+`.

Nurrl, 2018-03-07:
If it's printing `+`, I may have a fix, it could be a `const char*` -> `unsigned char` cast error.

Skyluro, 2019-01-08:
I had the same issue. Change the double quote to single

Keyboard.press("r");     //incorrect
Keyboard.press('r');      //correct