Dukweeno/Duckuino #68: Digispark

KasperSvendsen, 2018-05-23:
*Short introduction to your problem.*

I tried submitting code with the Arduino IDE to a Digispark. It gives me this error: 
Keyboard.h:25:17: fatal error: HID.h: No such file or directory
 #include "HID.h"

Looks like some compatibility issues with the digispark. Can you add a feature that enables support for this device?

__Input code:__
```vb
REM Author: Cody Theodore
REM Title: OSX Youtube Blast
REM This payload will open terminal, crank up the Macs volume all the way, then open a youtube video of
REM your choice by replacing the link.
DELAY 1000
GUI SPACE
STRING terminal
DELAY 500
ENTER
DELAY 4000
STRING osascript -e 'set volume 7'
DELAY 500
ENTER
DELAY 500
STRING open https://www.youtube.com/watch?v=dQw4w9WgXcQ
DELAY 500
ENTER
```

__Output code:__
```arduino
/*
 * Generated with <3 by Dckuino.js, an open source project !
 */

#include "Keyboard.h"

void typeKey(int key)
{
  Keyboard.press(key);
  delay(50);
  Keyboard.release(key);
}

/* Init function */
void setup()
{
  // Begining the Keyboard stream
  Keyboard.begin();

  // Wait 500ms
  delay(500);

  // Author: Cody Theodore
  // Title: OSX Youtube Blast
  // This payload will open terminal, crank up the Macs volume all the way, then open a youtube video of
  // your choice by replacing the link.
  delay(1000);

  Keyboard.press(KEY_LEFT_GUI);
  Keyboard.press(' ');
  Keyboard.releaseAll();

  Keyboard.print("terminal");

  delay(500);

  typeKey(KEY_RETURN);

  delay(4000);

  Keyboard.print("osascript -e 'set volume 7'");

  delay(500);

  typeKey(KEY_RETURN);

  delay(500);

  Keyboard.print("open https://www.youtube.com/watch?v=dQw4w9WgXcQ");

  delay(500);

  typeKey(KEY_RETURN);

  // Ending stream
  Keyboard.end();
}

/* Unused endless loop */
void loop() {}
```

__Error:__
```js
/* Any error outputted by the page or the console, 'none' otherwise */
```

Nurrl, 2018-05-23:
Hi @KasperSvendsen, to use the Digispark, you need to use the little down carret on the "Compile" button then click compile and download. It is supported.

Let me know if you have other issues after doing that.

Nurrl, 2018-05-23:
I'll close the issue, because I don't get your issue, feel free to send another message here if you need any support !

closed 2018-05-23 by Nurrl