Dukweeno/Duckuino #70: Error: No module loaded!

jvlavl, 2018-06-08:
Hello,

I have dowloaded duckuino to my local webserver and configured this virtual host to listen to port 81
When I open the site http://hostip:81 site is loading correctly.

pasting the ducky code in it the compile button lights up but when I hit it It gives me the error No module loaded!

How can I solve this?

__Input code:__
```vb
DELAY 3000
GUI r
DELAY 500
STRING notepad
DELAY 500
ENTER
DELAY 750
STRING Hello World!!!
ENTER
```

Rgds,

John

Nurrl, 2018-06-09:
Hi @jvlavl, I gonna need a screenshot please.

jvlavl, 2018-06-11:
Hello,

Here is the requested screenshot.
[image]

Thanks in advance

Nurrl, 2018-06-11:
Hi @jvlavl, have you selected a module with the caret on the compile button ?

jvlavl, 2018-06-11:
Hello,

That"s also part of the issue i Can 't select a module. Aldo they are present in the root directory of the site.

Just die a git globe info/var/www/ducky apache config is pointing to /var/www/Lucky

Rgds,

John

Nurrl, 2018-06-11:
Can you open the Javascript console and refresh + try to compile and send a screnshot of the console ?

Nurrl, 2018-06-11:
I think there is a problem of module loading ^^

jvlavl, 2018-06-12:
Hello,
Hereby the requested screenshots

[image]

[image]

Rgds,

John

Nurrl, 2018-06-12:
Hi @jvlavl, that's a non-issue, you can't access https://nurrl.github.io/LocaleKeyboard.js/locales/localeList with the client so the code breaks..
For now, Duckuino needs to access this file from internet to work properly.

Plazmaz, 2018-06-12:
@Nurrl won't that drive us to CORS issues?

Nurrl, 2018-06-12:
@Plazmaz CORS ?

Nurrl, 2018-06-12:
@Plazmaz Yeah, I got it, yes it will, I'm trying to merge the two projects, but now I don't have much time to do it..