Dukweeno/Duckuino #75: Need better instructions

Build-0-Matic, 2018-08-17:
Hi!
I'm trying to create a new locale for the keyboards used in the region but I'm running into a small problem... I just can't seem to figure out how to make this work.
It's my first time programming an Arduino as a keyboard...

So far, your library does print the string sequence on the screen, but some keys need to be remapped.
One of the hard things is that my keyboard uses combination keys for the French accents. Like when I want to print a ç, I need to press what the arduino thinks is ] and the c (I believe). And my # is located at the place the ~ is on the English American keyboard... and so on. So I need to re-write a whole bunch of key codes.

I'd really love to be able to create a new locale and share it...

Nurrl, 2018-08-19:
Hi @Build-0-Matic, you need to make your own locale file to map each key code to a character, unfortunately, I didn't found any other way to do it than trying and retrying until it works !
Also did you read this ? https://github.com/Nurrl/LocaleKeyboard.js/wiki/How-to-create-a-locale-%3F!

Nurrl, 2018-08-19:
Continuing there -> https://github.com/Nurrl/LocaleKeyboard.js/issues/50

closed 2018-08-19 by Nurrl