Dukweeno/Duckuino #81: Keyboard.press returns wrong values when using Canadian keyboard

Build-0-Matic, 2018-09-22:
I was just working on a project and discovered a new issue. When you send the `Keyboard.press("r") ;` command, instead of returning the r key it returns =

This is really weird since the keymap works fine for the `Keyboard.print(F( ));` function

Anybody got an idea on why this is?

Nurrl, 2018-09-22:
Hi @Build-0-Matic, by returning, you mean actually writing as if it was a keypress ? 
Can you provide an example, and also with what board are you trying to write these ?

Build-0-Matic, 2018-09-23:
Yes, I'm talking about keypresses.
Something weird has happened on github because I added a reply to my original message with more description and it has vanished... I'll just type it again.

I'm using an Arduino Pro Micro clone.
Normally, the script would press win-r to open the run box and then type some code I got from Hak5Daren's duckyscript repository and press enter.

But what happens is that the script presses win-= which opens up the magnifier then types in the code followed by nothing.

When I looked at the code, I noticed that the duckyscript converter doesn't recognize the `Keyboard.press("");` function when it contains any letter. Originally in the duckyscript version called for `GUI r` which the converter transforms to 
`Keyboard.press(KEY_LEFT_GUI);
Keyboard.press("r");
Keyboard.releaseAll();`
when it should be (according to my testing)
`  Keyboard.press(KEY_LEFT_GUI);
  Keyboard.print(F("r"));
  Keyboard.releaseAll();`

Also, it sends the `typeKey(KEY_RETURN);` function at the end which doesn't seem to work. I had to replace it on my script with 
`Keyboard.press(KEY_RETURN);
Keyboard.releaseAll();`
in order to make it work....

This is all just weird to me...

Nurrl, 2018-10-02:
That seems weird to me too :/ can you provide me the exact code + module you used to convert obtain these weird conversions errors ? Thanks @Build-0-Matic

Build-0-Matic, 2018-10-02:
No problem.
I used the code here https://github.com/hak5darren/USB-Rubber-Ducky/wiki/Payload---Fake-Update-screen
Which I input in the online duckduino converter (https://nurrl.github.io/Duckuino/)

Which in turn returns this
`/**
 * Made with Duckuino, an open-source project.
 * Check the license at 'https://github.com/Nurrl/Duckuino/blob/master/LICENSE'
 */

#include "Keyboard.h"

void typeKey(uint8_t key)
{
  Keyboard.press(key);
  delay(50);
  Keyboard.release(key);
}

/* Init function */
void setup()
{
  // Begining the Keyboard stream
  Keyboard.begin();

  // Wait 500ms
  delay(500);

  // fake update LUL
  // author: Judge2020 https://judge2020.com
  // edited by: DanielL99 on github (thanks!)
  // 
  // let the ducky HID enumerate
  delay(3000);
  Keyboard.press(KEY_LEFT_GUI);
  Keyboard.press("r");
  Keyboard.releaseAll();

  delay(500);
  Keyboard.print(F("iexplore -k http://fakeupdate.net/win10u/index.html"));

  typeKey(KEY_RETURN);

  // Ending stream
  Keyboard.end();
}

/* Unused endless loop */
void loop() {}`

I chose a small script to test it out so that it's not overwhelming, but the result puzzles me.

Nurrl, 2018-10-23:
The `"r"` over `F("r")` is definitely a Duckuino bug, the other bugs are either related to an incorrect timing or a bad Arduino or even a bug in the Arduino library.

Build-0-Matic, 2018-10-23:
Oh well... at least this is easy to fix before compiling on the Arduino, especially since most of the rest is OK.

Nurrl, 2019-04-29:
This is fixed in the `v2.1` branch, so I'll close the issue ! Thanks

closed 2019-04-29 by Nurrl

Nurrl, 2019-04-29:
Update: I merged v2.1 into master, you could just use the live version now !

Please open a new issue if you see any bug !