Dukweeno/Duckuino #82: Contribute new locale

moswed, 2018-09-29:
hi @Nurrl 

since you closed pull requests for the https://github.com/Nurrl/LocaleKeyboard.js repo, where can i put my created .lang file so it gets added to your converter? :)

Nurrl, 2018-10-02:
Hi @moswed, in order to create a new locale for the next version, you need to be on the v2.1 branch and look how I made others, it's very straight forward, but if you have any issue, please let me know :)
You'll need to place it in `modules/arduino/locales/`.
Thanks for you involvement in the project by the way :)

moswed, 2018-11-22:
somehow i missed your answer to this, sorry ;)
will include my created file asap and make a pull request.

And thank YOU for this nice project 💃 

hope to see the locale gets added to offical malduino script converter :)

closed 2018-11-22 by moswed