Dukweeno/Duckuino #92: QWERTY keyboard

ax3lt, 2019-04-20:
Hi I have a problem with the keyboard because it is set to the English AZERTY and I have the QWERTY and therefore the commands put them wrong .....
I have seen the other queries but I can't find them ....
the Download sketch button! only opens a window where nothing is selectable ...
You can help me?

MBR07, 2019-04-28:
There is an arrow right by side of "Download" on the website. Try it out!

ax3lt, 2019-04-28:
mmm Sorry but I can' t find the Arrow...
Is that near download sketch?
because if I click on it a window opens and I can only write I think of the links and do downloads but it doesn't work

MBR07, 2019-04-29:
Right beside download (not compile), you click it and there is a menu with
white text.

ax3lt, 2019-04-29:
yes I clicked it but a white screen opens up and I don't know what to do since I can't select anything

Nurrl, 2019-04-29:
Hello @ax3lt, this is fixed in the new version, I'm still working on it ! But you can switch to branch v2.1, download the project and use it !

closed 2019-04-29 by Nurrl