Dukweeno/Duckuino #93: GUI r doesn't open GUI r

MBR07, 2019-04-28:
## I. Describe your environment

- Platform:
Desktop
- OS:
Win10x64

## II. Describe your problem

**How to reproduce the bug**
Steps to reproduce the behavior:
1. Get an Arduino Pro Micro
2. Type "GUI r" in duckuino
3. Keyboard.press("r"); (It is supposed to output Keyboard.press('r');  )

**Observed behavior**
When trying to avhieve GUI r the program outputs the "wrong apostrophe" making the key not push.

**Expected behavior**
GUI r presses GUI + R

Nurrl, 2019-04-29:
Hello @MBR07, this is fixed in the new version, I'm still working on it ! But you can switch to branch `v2.1` and use it !

closed 2019-04-29 by Nurrl

Nurrl, 2019-04-29:
Update: I merged v2.1 into master, you could just use the live version now !

MBR07, 2019-05-02:
Merci!