Dukweeno/Duckuino #99: Arduino IDE Error

inventorman, 2019-08-22:
Hello,
Thanks for your awesome project.
I wonder if you could help me with this. After uploading your compiler code to my digispark board, I face with this:

Multiple libraries were found for "Keyboard.h"
In file included from C:\Users\xxx\AppData\Local\Temp\arduino_modified_sketch_539447\sketch_aug22a.ino:5:0:

 Used: D:\xxx\Arduino\IDE\arduino-1.8.9\libraries\Keyboard
D:\xxx\Arduino\IDE\arduino-1.8.9\libraries\Keyboard\src/Keyboard.h:25:17: fatal error: HID.h: No such file or directory

 Not used: C:\Users\xxxxxxxxx\Documents\Arduino\libraries\Keyboard-master
 #include "HID.h"

Nurrl, 2019-08-24:
Hi @inventorman, 

Did you selected the good board in the Arduino IDE ?

inventorman, 2019-08-25:
> Hi @inventorman,
> 
> Did you selected the good board in the Arduino IDE ?

Yes, and I'm sure about it because I can upload anything properly except for the code which includes "Keyboard.h".

inventorman, 2019-08-25:
Here is a sample code. I wonder if anyone can test it. I appreciate it...

`#include "Keyboard.h"

void typeKey(int key)
{
  Keyboard.press(key);
  delay(50);
  Keyboard.release(key);
}

/* Init function */
void setup()
{
  // Begining the Keyboard stream
  Keyboard.begin();

  // Wait 500ms
  delay(500);

  delay(500);

  Keyboard.press(KEY_LEFT_GUI);
  Keyboard.press('r');
  Keyboard.releaseAll();

  delay(300);

  Keyboard.print("notepad.exe");

  delay(300);

  typeKey(KEY_RETURN);

  delay(300);

  Keyboard.print("Hello World!");

  // Ending stream
  Keyboard.end();
}

/* Unused endless loop */
void loop() {}`

Nurrl, 2019-11-22:
Hi @inventorman, retry by using the good module, the *Digispark* module.
Click the little down arrow next to the *Compile* button, and select Digispark, it should work, if not, please post another issue.

closed 2019-11-22 by Nurrl