Dukweeno/Duckuino #103: Keyboard not fount

lidlman1230, 2019-10-24:
Are you using the live version that could be found [here](https://nurrl.github.io/Duckuino/) ?
  - [ ] Yes, absolutely
  - [ ] No, I might try it out

What type of issue is it ?
  - [ ] Graphical issue (UX, UI, Graphical Glitch)
  - [ ] Feature issue (Command not/bad implemented, Non functionnal feature)
  - [ ] Crash (The buttons clicks does nothing at all, Error in console)
  - [ ] Question (Any type about the project :D)

```
Here are relevant errors/infos outputed by the code !
```

_Describe your problem here :D_

Arduino: 1.8.10 (Windows 10), Board: "Arduino/Genuino Uno"

In file included from C:\Users\hohoh\Documents\Arduino\sketch_oct24a\sketch_oct24a.ino:5:0:

C:\Program Files (x86)\Arduino\libraries\Keyboard\src/Keyboard.h:29:2: warning: #warning "Using legacy HID core (non pluggable)" [-Wcpp]

 #warning "Using legacy HID core (non pluggable)"

  ^~~~~~~

C:\Users\hohoh\Documents\Arduino\sketch_oct24a\sketch_oct24a.ino: In function 'void typeKey(int)':

sketch_oct24a:9:3: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
   Keyboard.press(key);

   ^~~~~~~~

C:\Users\hohoh\Documents\Arduino\sketch_oct24a\sketch_oct24a.ino: In function 'void setup()':

sketch_oct24a:18:3: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
   Keyboard.begin();

   ^~~~~~~~

sketch_oct24a:25:18: error: 'KEY_LEFT_CTRL' was not declared in this scope

   Keyboard.press(KEY_LEFT_CTRL);

                  ^~~~~~~~~~~~~

sketch_oct24a:26:18: error: 'KEY_LEFT_ESC' was not declared in this scope

   Keyboard.press(KEY_LEFT_ESC);

                  ^~~~~~~~~~~~

sketch_oct24a:35:11: error: 'KEY_RETURN' was not declared in this scope

   typeKey(KEY_RETURN);

           ^~~~~~~~~~

sketch_oct24a:39:11: error: 'KEY_TAB' was not declared in this scope

   typeKey(KEY_TAB);

           ^~~~~~~

sketch_oct24a:59:18: error: 'KEY_LEFT_ALT' was not declared in this scope

   Keyboard.press(KEY_LEFT_ALT);

                  ^~~~~~~~~~~~

Multiple libraries were found for "Keyboard.h"
 Used: C:\Program
Multiple libraries were found for "HID.h"
 Used: C:\Program
exit status 1
'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Nurrl, 2019-10-27:
Could you provide the .zip *Duckuino* produced ?

Nurrl, 2019-10-27:
Also you can't use Arduino Uno with this project !!

lidlman1230, 2019-10-28:
Okay i set the wrong Arduino

closed 2019-10-28 by lidlman1230