Dukweeno/Duckuino #110: HID.h: No such file or directory

keldnorman, 2020-06-25:
## I. Describe your environment

- Platform: Laptop Windows 10
- Arduino Version: 1.8.3 (latest)

## II. Describe your problem

- cloned the code (downloaded the zip file) 
- double clicked the Sketch.ino file
- selected tools -> port -> arduino/genuino micro (com9)
- compile failed with the error: HID.h: No such file or directory

Pressed compile button...

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\xxx\AppData\Local\Arduino15\packages -tools C:\Program Files ....
...
DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for HID.h: []
ResolveLibrary(HID.h)
  -> candidates: []
In file included from C:\Users\xxx\Desktop\BadUSB_ATMEGA32U4\Sketch\Sketch.ino:6:0:
Keyboard.h:25:17: fatal error: HID.h: No such file or directory
 #include "HID.h"
                 ^
compilation terminated.
exit status 1
HID.h: No such file or directory

keldnorman, 2020-06-25:
Fixed with :   select tools -> board -> Arduino Leonardo

closed 2020-06-25 by keldnorman

Nurrl, 2020-06-25:
Hi @keldnorman, yes unfortunately some Arduino boards don't have Keyboard Emulation capabilities